drolsen/DRO-Matic #45: Peristaltic Pump Setup

SeriousBuds, 2016-12-22:
I'd love to build this setup and document my whole experience on rollitup.org, where you'll get a lot of interested minds in the project - but the syringe pump method is a big barrier for me.. do you plan on releasing a peristaltic pump guide? 

Thanks for your hard work on this, can't wait to see it when it's finished.

drolsen, 2016-12-24:
Hi SeriousBuds, thanks and that would be awesome!
To answer your question, yes I do had done. Currently the code is not finished as you might have noticed as I've been inundated with family events and holidays plus full time work. But, under the configuration of the system, you will have the option to set fixed sizes on each channel. It is here that the system will know if you are using a peristaltic pump or not and adjust its dosing logic accordingly. So again, fixed channel size = syringe based pumps, where is no fixed size = continues or rather peristaltic pumps. And again, the only difference really between the two is that fixed sizes have to stop and push solution out before meeting desired dose size, while non-fixed sized pumps do not have to stop and just keep going till desired solution amount is met.

That said I've been trying (with the few moments I've had since building this) to get around my final issue of my channel session churning. You see the Arduino only has a single core, thus multi threading is not possible. So when the system needs to be going over each channel's available configured sessions and making a determination on if to dose or not, you are locked out of the system menus for a full 20 seconds or more passed (based on your configure, mine are large). That was not acceptable to me, and I ventured down many roads to solving this. One of which included having the channel's session's loaded into memory via a "loading sessions" screen after building a crop or turing on system to pre-built crop. This got tricky as now you were editing both SD crop config files/session data AND data loaded into memory.. also did I mention how finite arduno memory is hehe.

In the end as of this evening, I've finally figured out my best solution that gets me back on track and can finally wrap beta of this thing up and record all the features I've said I would do for months now. It does mean that in some circumstances when pressing a arrow key on the key pad to get menus up, you might have to wait 2 seconds, while in other instances it will be 0 second wait. I can live with this and might just make a universal number of 2 seconds always to bring menus up to make it more consistent.

Lastly, again cause of the system single core limitation we can't stop a dosing once it has started and the user will have to cut power if needed in a emergency (nothing I can do about this at all).

So with that I say Merry Christmas sir, and I will be merging development branch into master in the next few days. In the next few weeks I plan on purchasing a single peristaltic pump to test the non-fixed configure against the OS as well... unless you have one already :)

drolsen, 2017-01-04:
Hi again SeriousBuds, I just wanted to give you the heads up that I've just ordered a single 12v Perstaltic Pump today and should be arriving friday. I should have my OS confirmed working with these pumps by the end of the weekend.

https://www.amazon.com/Dosing-Peristaltic-Liquid-Aquarium-Analytical/dp/B01HRPKB0O/ref=sr_1_1?ie=UTF8&qid=1483557388&sr=8-1-spons&keywords=peristaltic+pump&psc=1

Also, I've just put in my order for two electric 1" ball valves and I will be configuring an entire top off/ drainage system to the OS by mid next week or early next weekend.

I do need to point out that you will still need to purchase an Arduino mega, SD card reader, RealTime Clock chip, wires and a breadboard.. but putting that together is going to cost very little and take almost no time to setup. I will be making a video on the setup of these pump types in about a week.

drolsen, 2017-01-08:
Update: My peristaltic pump arrived yesterday and I got it hooked up to the Arduino as channel 1 of relay (bypassing the chips and breadboard wiring)!! 

Since then I've finished up extending out the channel size screen to allow you to set ∞ (infinity or rather continues pump type) or still a fixed size. If you set it to ∞ for peristaltic pump, the channel calibration screen allows you to not set number of rotations, but instead ml per minutes (per your peristaltic pump manufacture spec). Finally I'm using this the channel size to again condition how we check channel sessions per min and do the pumping logic.

With that, I can confirm that these pump types now work with the OS.
I will push this code out later today once I get a few more things cleaned up. Then all that is left is for me to create a start to finish video on how to build a doser using 10 of these peristaltic pump and this OS.

I will be re-ordering (on 15th of this month) a new arduino mega 2560, keysheild, realtime clock chip, SD card reader chip, 16 channel relay, 9 peristaltic pumps, pH reader, EC reader to create this video for you and others so my goal will be to try and cut costs as much as possible vs. my current fixed pump build. Also I don't have a use for two dosers, so I might be doing a give away on my youtube.com channel of this cheaper build once done and video is out.

Thanks!

closed 2017-01-08 by drolsen