drolsen/DRO-Matic #55: Build OS Configure Web Interface

drolsen, 2017-01-12:
Approaching close to the finish of being able to configure all OS settings from LCD shield, it is now time to consider building a quick web interface tool that allows you to configure a crop via interfacing with calendar and events as sessions.

Tool can be hosted on jsfiddle and embedded and linked to with revisions available for later on and requiring no hosting. Here are the found resources for this app thus far:

- jQuery latest (calendar plugin requirement)
- FullCalendar http://fullcalendar.io/
- JSZip https://stuk.github.io/jszip/

User Experience Outline:

- No login required
- Fills out crop name
- Fills out channel number
- Gets presented with entire interface
- Interface is essentially a large month to month calendar with options above it.
- User can switch what channel and what session they are under and use calendar to configure session times.
- User can add and remove sessions while under any of the channels they configured in the beginning.
- User can also alter number of channels at any point by max is 10 min is 1.
- All repeat, amount, and start time for  sessions are set in the events of the calendar.
- Once done, user can hit generate button and a zip file / download will be given to them with crop fully configured. User can take contents of zip and put on SD card and plug into Arduino/OS and load crop up.

bigjuicy79, 2017-10-14:
Hey I'm new to Arduino and I was wondering if you could email me what codes I need to add to my scetch or if you have a better way to upload all this to my Arduino email is bigjuicy79@gmail.com

stvnmbr1, 2019-09-16:
so... is this still being developed?

bjrush, 2019-09-17:
That's my question. Nobody will  reply to my questions.

stvnmbr1, 2019-09-17:
I think development just stopped on this project ..

bjrush, 2019-09-17:
And I've been trying to just wire it up to schematics he furnished on line and am confused when other people comment why this and why that. I'm installing my greenhouse in Dec and would like the hands free idea but everything has stopped on the info or answering anything about this project.

iamhollis84, 2021-03-22:
> That's my question. Nobody will  reply to my questions. 
> 
> Sent from Yahoo Mail on Android 
>  
> so... is this still being developed?

I'm not expert but I've been working on fixing a lot of bugs for the last few hours. What are you stuck on?