drolsen/DRO-Matic #100: No Wire.endTransmission() needed after Wire.requestFrom()

Koepel, 2017-08-13:
Hi, do you use a special Wire library ? I noticed odd usage of the Wire library.

After a Wire.requestFrom() you use Wire.endTransmission(). That is not needed. The Wire.requestFrom() is a complete I2C transaction on its own. The Wire.endTransmission() is only used when writing data.

I suggest to keep the "Wire.beginTransmission - Wire.write - Wire.endTransmission" close together. You have the Wire.beginTransmission() in a seperate function, which is unusual. In the Arduino Wire library, the I2C bus transaction is fully done with Wire.endTransmission(). The other two functions set variables and write to a buffer.

You request 48 bytes with Wire.requestFrom(), but the Arduino Wire library has a maximum of 32 bytes.

drolsen, 2017-08-13:
`Hi, do you use a special Wire library ? I noticed odd usage of the Wire library.`
Nope normal Wire lib.

`After a Wire.requestFrom() you use Wire.endTransmission(). That is not needed. The Wire.requestFrom() is a complete I2C transaction on its own. The Wire.endTransmission() is only used when writing data.`
This code comes prescribed from the Atlas Scientific Tentacle Shield example for Arduino continues i2c found here:
https://github.com/whitebox-labs/tentacle-examples/blob/master/arduino/continuous/i2c_continuous/i2c_continuous.ino

I assume they recommend the manual release of the requestFrom for legacy purposes in case someone doesn't have Arduino 1.0.1 and nothing more:
`As of Arduino 1.0.1, requestFrom() accepts a boolean argument changing its behavior for compatibility with certain I2C devices.

If true, requestFrom() sends a stop message after the request, releasing the I2C bus.

If false, requestFrom() sends a restart message after the request. The bus will not be released, which prevents another master device from requesting between messages. This allows one master device to send multiple requests while in control.

The default value is true.`
However you're right, I've drop this from my code cause it's just crazy pants to try and backwards compatible something that was releases back in 2012 for code sakes. 

`I suggest to keep the "Wire.beginTransmission - Wire.write - Wire.endTransmission" close together. You have the Wire.beginTransmission() in a seperate function, which is unusual. In the Arduino Wire library, the I2C bus transaction is fully done with Wire.endTransmission(). The other two functions set variables and write to a buffer.`

Long story short is that I migrated away from an alternative pH and EC probe manufacture, got frustrated with trying to remember my true i2c ids doing event based logic build out. Built this method to pass abstracted 0-3 bytes that were easier to remember than true i2c ids. This however was never needed as I've rolled up the channel array ids into human readable definitions instead. You can also see hints of my frustration because I split out two separate methods for getting pH or EC probe readings even though they are virtually the same method. This is cause I got sick of trying to read my logic in bytes alone, so now I can ready my logic by method name and byte. The openWaterProbeChannel() method has been removed and Wire's begin / end transition are now within the same method(s).

`You request 48 bytes with Wire.requestFrom(), but the Arduino Wire library has a maximum of 32 bytes.`
Gasp, I've lowered this down to 32. Thanks!

closed 2017-08-13 by drolsen

Koepel, 2017-08-13:
Thanks for your answers, I will make an Issue at whitebox-labs. 

The Wire.endTransmission() is used to write data after a Wire.beginTransmission() and Wire.write(), it can not be used to release the bus. Well, it can at this moment for the Wire library for AVR chips, but that is pure luck and not in the documentation/reference.