drolsen/DRO-Matic #113: Top-off correction need to be both time and volume based.

drolsen, 2017-08-20:
Right now, EC correction does two steps: 
- drains plant water by configured amount (gallons) converted into time (seconds) based on out flow meter
- feeds top-off water from reservoir to plants by (gallons) converted into time (seconds) based on out flow meter

However the problem here is that water pressure / flow rate from reservoir changes as reservoir gets lower in water volume. Because of this, plant water levels can get too low during top-off cycles leaving roots exposed. 

Instead of feeding of top-off water based on time, we should use out flow meter and measure exact gallon volume. This will keep water levels for plants more constant over the life of a regimen.

maximillia007, 2017-08-21:
Can you use a level sensor and fill until the sensor recognizes the proper level?

drolsen, 2017-08-25:
Yes one could, but I would prefer not to make anymore modifications to the cabinet just yet :)
Right now with my local code, I'm able to keep levels darn near perfect now with only about 1/4 of a gallon (out of 28 gallons) slipping. So I'm pretty sure this will be a suitable approach.

If I was going to add any more float level sensors to a setup, they would be for isolated systems that turn on or off my water pumps based on water level so they never run dry.

Great to see people asking questions!!

admwkd, 2017-09-13:
I use a float valve like in an evaporative cooler simple and it keeps it topped off all the time.

drolsen, 2018-01-25:
This is no longer an issue having further beta tested the unit in actual production.
Closing issue.

closed 2018-01-25 by drolsen