drolsen/DRO-Matic #115: Alternate Probes

H20frks, 2018-01-25:
Mr. Olsen this project is fantastic and want to look at building this for my Hydroponics garden outside, but the Atlas probes and boards are pretty expensive. What would it take to have the OS compatible with other EC/PH probes (couple examples below)? Do you have an active forum for this project? 
[PH Probe: ](https://www.dfrobot.com/product-1025.html)
[EC Probe:] (https://www.dfrobot.com/product-1123.html?gclid=Cj0KCQiA-qDTBRD-ARIsAJ_10yKWDGon203htZ6dMRXqX6cp960XcQ1lZjxinw7uSXDezrHRrVK_bPEaApgbEALw_wcB)

drolsen, 2018-01-25:
Hi @H20frks and thanks!
I understand cost is a huge issue for most right now building these. The reason the Atlas board is so expensive is cause it a fully power isolated. This means you can have multiple probes within the same water source and no current between the probes will be created. Unfortunately I don't think that dfrobot board there is power isolated and will work for this application. That said, there are cheaper probes and power isolating chips for doing this that are not Atlas coming out as of recent, but to re-write the OS to work with them might be a little difficult right now. I will consider writing some alternative probe setups for the open source version here soon, of if there are some forks people want to contribute I'm willing to include in this repo.

No I don't have any forum open right now unfortunately, but I am going to be putting up a website later this spring after I get some PCB work done. I am about 6-8 weeks out right now from having my own physical circuit board being manufactured for DRO-Matic vs. buying all these parts and components. I going to start selling these boards on my website by this summer. My board is going to cost substantially less then my build list here for the open source project, so please stay tuned.

Here is a video of the open source version in action:
[https://www.youtube.com/watch?v=52KLw9LHk5s](https://www.youtube.com/watch?v=52KLw9LHk5s)

H20frks, 2018-01-26:
Thank you for the quick response and all your hard work making the project. I am looking forward to your further progress. I have a home automation controller that I would love to figure how how to post data to the server and build out graphs for further analyzing one crop settings from another. This is fantastic work. 

Thanks again!

mercdev, 2018-09-20:
> I going to start selling these boards on my website by this summer. My board is going to cost substantially less then my build list here for the open source project, so please stay tuned.
> 

How's progress on this?

baylanger, 2020-08-29:
DFROBOT - PH & EC probes will not work with this project.  None of those probes are good for 7x24 operations, please go check the specs and cross check data.  The EC and PH probes can't stay in the solution for a long period of time.  The PH probe doesn't last long either.  AFAIK DRO-Matic needs 7x24 at least for current solution in the system - if I am wrong, let me know.

Could DRO-Matic support the following? @drolsen

- 7x24  OR  drop-in/out cheaper  probes
- support to run the system on a single pair of probes

Reducing the overall cost of the system and increase life expectancy of all probes be 7x24 or not.
For people often home, can we allow the system to work with a single pair of ph/ec probes?

IF using a single pair to fill the mixing-tank, the operator must put the probes in its place and then tell DRO-Matic to launch that sequence.  Once mixing is done the probes can be removed.

Here's the logic to adjust the current solution, I have never used the system so please adjust accordingly.

{noformat}
- IF 1 probe is detected in solution (spot check!)
  => LCD 1st row : display and update value as long as probe in solution 

- IF the 2 probes are detected in solution
  => LCD 1st row : display and update both values
  => LCD 2nd row : Hold button to re-adjust solution
  => IF both probes not in solution => return

- IF hold button >= 2 seconds
  => LCD 2nd row : Safely remove your probes
  => function to adjust solution kicks in
{noformat}

I personally prefer the a 4 probes 7x24 system but when I'm home, the system should work if the operator removes temporaly 1 or multiple probes and resume normal whenever they are back.  I bet somehow DRO-Matic likely works like this but I just don't know yet.