drolsen/DRO-Matic #116: Failing on Build

H20frks, 2018-01-25:
I have VMMicro for VS and the build is failing time.h file is missing (not sure if the name should be timers.h. I downloaded the current master branch. Can you please help identify what I am doing wrong. 

Compiling debug version of 'DROMatic' for 'Arduino Nano w/ ATmega328'
 
DROMatic.ino:8: In file included from
 
ctime:51: fatal error  time.h  No such file or directory
   #include <time.h>
   compilation terminated
 
Crops.cpp:14: In file included from
 
Screens.h: 14:33: error: '>>' should be '> >' within a nested template argument list
   extern vector<vector<vector<byte>>> matrix
 
Crops.cpp: In function void setCropData(ArduinoJson::JsonObject&, bool)
 
Crops.cpp: 27:55: error: default argument given for parameter 2 of 'void setCropData(ArduinoJson::JsonObject&, bool)' [-fpermissive]
   void setCropData(JsonObject& d, bool returnHome = true){
 
Crops.cpp:8: In file included from
 
Crops.h: 17:13: error: after previous specification in 'void setCropData(ArduinoJson::JsonObject&, bool)' [-fpermissive]
   extern void setCropData(JsonObject& d, bool returnHome = true)
 
Crops.cpp: In function void printStatus(int)
 
Crops.cpp: 481:29: error: default argument given for parameter 1 of 'void printStatus(int)' [-fpermissive]
   void printStatus(int dir = 0){
 
Crops.cpp:8: In file included from
 
Crops.h: 45:13: error: after previous specification in 'void printStatus(int)' [-fpermissive]
   extern void printStatus(int dir = 0)
 
Crops.cpp: In function void printPHAmount(int)
 
Crops.cpp: 493:31: error: default argument given for parameter 1 of 'void printPHAmount(int)' [-fpermissive]
   void printPHAmount(int dir = 0){
 
Crops.cpp:8: In file included from
 
Crops.h: 46:13: error: after previous specification in 'void printPHAmount(int)' [-fpermissive]
   extern void printPHAmount(int dir = 0)
 
Crops.cpp: In function void printPHDelay(int)
 
Crops.cpp: 522:30: error: default argument given for parameter 1 of 'void printPHDelay(int)' [-fpermissive]
   void printPHDelay(int dir = 0){
 
Crops.cpp:8: In file included from
 
Crops.h: 47:13: error: after previous specification in 'void printPHDelay(int)' [-fpermissive]
   extern void printPHDelay(int dir = 0)
 
Crops.cpp: In function void printOpen(int)
 
Crops.cpp: 660:27: error: default argument given for parameter 1 of 'void printOpen(int)' [-fpermissive]
   void printOpen(int dir = 0){
 
Error compiling project sources
Crops.cpp:8: In file included from
 
Crops.h: 52:13: error: after previous specification in 'void printOpen(int)' [-fpermissive]
   extern void printOpen(int dir = 0)
Debug build failed for project 'DROMatic'
 
Crops.cpp: 683:50: error: default argument given for parameter 2 of 'void printECCalibrations(String, int)' [-fpermissive]
   void printECCalibrations(String type, int dir = 0){
 
Crops.cpp:8: In file included from
 
Crops.h: 50:13: error: after previous specification in 'void printECCalibrations(String, int)' [-fpermissive]
   extern void printECCalibrations(String type, int dir = 0)

H20frks, 2018-01-26:
I am still having issues getting the project to build. Below is the output showing Timer.h no file. Do I need to download this library? I tried downloading the dev and master branches with the same results. I am also seeing an issue in the Screens.h.

Please advise.

Compiling 'DROMatic' for 'Arduino/Genuino Uno'
 
DROMatic.ino:11: In file included from
 
ctime:51: fatal error  Timer.h  No such file or directory
   #include <Timer.h>
   compilation terminated
 
Core.cpp:13: In file included from
 
Screens.h: 14:33: error: '>>' should be '> >' within a nested template argument list
   extern vector<vector<vector<byte>>> matrix
Error compiling project sources
Build failed for project 'DROMatic'

drolsen, 2018-01-27:
Hi @H20frks.
Sorry about the build issues you're having!
About 5 months ago I moved all the source files into the folder /src and renamed a few files, one of them being Timer.h into Timers.h. 
I'm not sure why your local copy isn't picking these changes up, but at the top of a number of the .cpp files:
- https://github.com/drolsen/DRO-Matic/blob/b7ac9b51f5ea2cd720a553f621835a865264a83e/DROMatic.ino#L27
- https://github.com/drolsen/DRO-Matic/blob/5e32c69a05349ba07c9d64f0a5a989d85bf5f489/src/Core.cpp#L16
- https://github.com/drolsen/DRO-Matic/blob/0cfb991f9839592938e216ac4a88b8b85d7af96f/src/Screens.cpp#L14
- https://github.com/drolsen/DRO-Matic/blob/5e32c69a05349ba07c9d64f0a5a989d85bf5f489/src/Crops.cpp#L16
- https://github.com/drolsen/DRO-Matic/blob/0cfb991f9839592938e216ac4a88b8b85d7af96f/src/Timers.cpp#L8

Make sure they point to Timers.h not Timer.h.
Let me know if this gets you any further.

H20frks, 2018-01-27:
Devin thanks for details. Not sure why I am not able to compile this. I have 

1. uninstalled VS
2. Re-Installed VS
3. Tried compiling the code with Arduino IDE
4. Manually add the includes
5. Un-installed Arduino IDE
6. Re-Install Arduino IDE
7. Tried to fork your project into VS and build still no luck.
Something is wrong and I cam not able to locate the issue.

drolsen, 2018-01-27:
You are using <src/Timers.h>?
You must of edited this include, cause its suppose to be quotes “src/Timers.h“ not <src/Timers.h>.

Brackets tell VS to go search for this include as a lib in an external location on this machine, where as quotes are relative to project.

I listed each file in my comment above and how they should be including the Times.h please take a look and adjust any edits you’ve made accordingly.

I’d also recommend backing up what you have now to a new location, downloading fresh copy of master and try compiling it instead.

H20frks, 2018-01-28:
Devin, I deleted all code and downloaded a fresh copy and I am still unable to compile the project in VS or Arduino IDE. I opened in both IDE sketch examples of Blink and compiled just fine and a more larger sketch that added libraries from MySensors.org and was able to compile these sketches with no issues. 
I am not understanding why I am not able to compile yours code. 

In VS I click Files->Open->ArduinoProj point to the .ino file.

H20frks, 2018-01-31:
Devin can you tell me what version of the Arduino you are using with this code?

H20frks, 2018-01-31:
I am not sure if I am getting closer to resolving this or further down the rabbit hole :-) 
I am no longer getting the library reference and am getting the below error for it looks like every function created in the project.
`Core.cpp.o (symbol from plugin)*: In function RelayToggle(int, bool)
(.text+0x0)*: multiple definition of RelayToggle(int, bool)
Core.cpp.o (symbol from plugin)*: (.text+0x0): first defined here`

I am using VisualStudio 2017 Community
Arduino 1.8.5

Attached is the full output. Any thoughts would be welcome.
[FaildBuild_Output.txt](https://github.com/drolsen/DRO-Matic/files/1680367/FaildBuild_Output.txt)

drolsen, 2018-01-31:
Hi @H20frks 
Sorry for delay, been a long work week already.

Here is my VS arduino version:
[image]

Do you have a project setup like mine here? 
If so with the project name selected, try going to **properties**:
[image]

Under this properties window, check **C/C++** rollout's **"Additional Include Directories"** and make sure **"$(ProjectDir)..\DROMatic;"** (or your project name) is included here so it will know where to include from.
[image]

Make sure VMicro's compiler is set to deep search:
[image]

Also if VS isn't picking up on the provided [libraries](https://github.com/drolsen/DRO-Matic/tree/master/libraries) you might need to manually move them over to the real Arduino IDE install location's libraries folder and restart VS. Mine arduino IDE is installed at **C:\Program Files (x86)\Arduino\libraries**.

I've never experienced that attached log, not entirely sure.
Hope this is helpful man.

PR #117 (2018-01-25): Fixes Timer.h and references that changed since new /src/ folder
Fixes #116
closed 2018-02-07 by drolsen