drolsen/DRO-Matic #123: Won't Compile on newest 1.8.7 Arduino IDE due to StandardCplusplus

joshvette001, 2018-10-25:
Project won't compile if using the newest Adruino IDE 1.8.7 due to some breaking changes made in Arduino 1.8.6 - avr-gcc 4.9.2 to avr-gcc 5.4.0 it causes StandardCplusplus not to work anymore and throw errors regarding (class std::basic_istream) and ostream. 

The easiest fix I have found currently without digging into this further is to simply install the 1.8.5 IDE which still uses avr-gcc-4.9.2

cqnguyen80, 2019-03-30:
> Project won't compile if using the newest Adruino IDE 1.8.7 due to some breaking changes made in Arduino 1.8.6 - avr-gcc 4.9.2 to avr-gcc 5.4.0 it causes StandardCplusplus not to work anymore and [...]

Try it with 1.8.5 and still wont compile

abufahad-1, 2019-03-30:
Try this 

https://github.com/drolsen/DRO-Matic/pull/126

cqnguyen80, 2019-03-31:
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\cqnng\Documents\Arduino\libraries\DROMatic\DROMatic.ino:14:0:

C:\Program Files (x86)\Arduino\libraries\StandardCplusplus/ctime:51:18: fatal error: time.h: No such file or directory

 #include <time.h>

                  ^

compilation terminated.

Multiple libraries were found for "SPI.h"
 Used: C:\Users\cqnng\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\libraries\SPI
 Not used: C:\Program Files (x86)\Arduino\libraries\SPI
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

cqnguyen80, 2019-04-08:
Got it to compile with Visuals Studio but for some reason I'm stuck on Crop Name screen with only forward  and reset button working

nilusha1993, 2021-02-13:
same, if anyone has fix it would be great service

nilusha1993, 2021-02-14:
> same, if anyone has fix it would be great service

actually got it all working perfectly now, took a couple of days and some editing in the code, if anyone needs any help how to get it all working let me know :) Cheers!

iamhollis84, 2021-03-18:
Can anyone help me get this up & running? It won’t compile

iamhollis84, 2021-03-22:
> > same, if anyone has fix it would be great service
> 
> 
> 
> actually got it all working perfectly now, took a couple of days and some editing in the code, if anyone needs any help how to get it all working let me know :) Cheers! 

I need help

nilusha1993, 2021-05-27:
use arduino 1.8.5