drolsen/DRO-Matic #124: 8 dosing pump for nutrient

abufahad-1, 2018-11-14:
Dear Devin

I am a beginner using Arduino
and I was searching for dosing controller for my hydroponic
I have found your project, which I am looking for.
But I have a question,
your project has 10 dosing pumps
8 for nutrient and 2 for PH.

What I do not understand, why 8 dosing pumps for nutrient?

We use A and B nutrient solution, so we need only 2 nutrient dosing pumps
in this case, if I want to use your project,
Is there a way to modify the code? To operate 4 pumps, 2 for PH, and 2 for A & B nutrient ?
If this is possible, is there anyway to edit?

Thanks and Regards

shotaatosh, 2018-11-14:
If you watched a video there are GH nutrients which come in 10+ components, three main part nutrients and other "add on's" if I can say so. 
Plus form my experience you need only one pump for PH, because it only goes in one way up or down (mostly up).
So in fact you need 3 pump automation, sol. A sol. B and PH down. if it is possible to modify the code like that I would love to see that and try it in my basement.

stepalexmuz, 2018-11-14:
there will be needs in PH Down OR PH Up depending of NH4/NO3
also it is possible to get the stable PH with NH4/NO3 about 1/16

abufahad-1, 2018-11-14:
shotaatosh
ok I understand what you say
Did not study programming, self-learn through online
If possible, let me know which files should be modified
And I will do the rest
But still I need ph up and down

abufahad-1, 2018-11-14:
In fact, I want to work this system for simple farmers
I will distribute this system to them free of charge, and teach them about hydroponics
So I hope I can modify this system for them

closed 2018-11-22 by abufahad-1