drolsen/DRO-Matic #128: Is this dead?

Jacoby6000, 2020-05-27:
Has development moved somewhere else? Is there a good alternative? I came across this when I was thinking about making something similar myself, but would much rather find something existing

bjrush, 2020-06-16:
I also have a very big interest in this but can't find help. If you find out more about this system please let me know. Joe G.

berten-ernie420, 2020-06-25:
I reached out to the author a month ago, he said to expect something new to be released this month, but that was before coronavirus happened. Maybe he will still release either an updated version or a new sketch entirely just later than anticipated. But I've been trying to get this version of the program debugged and edited for soil applocation for months with very little knowledge of C language coding. No luck so far. Ive found most of the issues come from outdated library language.

bjrush, 2020-06-27:
Thanks for the reply. I'm off work until September 1st and have a lot of time to devote to this. Wiring has been my downfall. Do you have anything that I can use for diagram?

gromyjoe, 2020-06-27:
I have a request if this is still active.
It would be nice to have this mqtt enabled so we can add it to the IOT

baylanger, 2020-08-29:
For MQTT , I see you might need to create a "Arduino IoT account" but I am not positive if this is a hard requirement.  Does Arduino MQTT requires such account or not?

I looked at this project many years ago and I just started to look at it again just today to realize it seems day but glad to hear author seems to be on something to be published before end of 2020 🤞

Would a REST API that allows to read sensors' value and make config changes be of interest?

How about pushing metrics from sensors to postgres, seems there is such a library available and Grafana to create dashboards?

johnny6uitar, 2020-11-26:
Hey I wired the entire thing, it took me about 30-40 hours to make it.  You may want to separate the high voltage stuff from the lower voltage stuff in a seperate waterproof container.  Problem is I cannot get the code to compile either.