drolsen/DRO-Matic #129: Please help this crop error

nilusha1993, 2021-02-13:
Hi, i got the code to compile and upload successfully with few warnings, but the lcd sheild is at Crop name and it does not let me input alphabet, only the forward button works and the back button has become the top bottom. other buttons dont work. using dfrobot lcd sheild 2.0

these are the errors would be great help if anyone can help me through this 

using arduino ide 1.8.5 and same libraries as drolsen.

warnings as follows - 

sketch\Crops.cpp:507:14: warning: character constant too long for its type

  if (type == 'low'){

              ^

sketch\Crops.cpp:511:14: warning: character constant too long for its type

  if (type == 'mid'){

              ^

sketch\Crops.cpp:515:14: warning: character constant too long for its type

  if (type == 'high'){

              ^

sketch\Crops.cpp:524:14: warning: character constant too long for its type

  if (type == 'dry'){

              ^

sketch\Crops.cpp:528:14: warning: character constant too long for its type

  if (type == 'low'){

              ^

sketch\Crops.cpp:532:14: warning: character constant too long for its type

  if (type == 'high'){

              ^

sketch\Crops.cpp: In function 'void setCropData(ArduinoJson::JsonObject&, bool)':

sketch\Crops.cpp:83:55: warning: default argument given for parameter 2 of 'void setCropData(ArduinoJson::JsonObject&, bool)' [-fpermissive]

 void setCropData(JsonObject& d, bool returnHome = true){

                                                       ^

sketch\Crops.cpp:24:13: note: previous specification in 'void setCropData(ArduinoJson::JsonObject&, bool)' here

 extern void setCropData(JsonObject& d, bool returnHome = true);

             ^

sketch\Crops.cpp: In function 'void cropRename(int)':

sketch\Crops.cpp:147:13: warning: NULL used in arithmetic [-Wpointer-arith]

  if (dir != NULL){

             ^

sketch\Crops.cpp: In function 'void printStatus(int)':

sketch\Crops.cpp:540:29: warning: default argument given for parameter 1 of 'void printStatus(int)' [-fpermissive]

[... 767 lines cut ...]

C:\Users\nilus\Documents\Arduino\libraries\DS3231\DS3231.cpp:291:106: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\nilus\Documents\Arduino\libraries\DS3231\DS3231.cpp:291:106: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\nilus\Documents\Arduino\libraries\DS3231\DS3231.cpp:291:106: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\nilus\Documents\Arduino\libraries\DS3231\DS3231.cpp:291:106: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\nilus\Documents\Arduino\libraries\DS3231\DS3231.cpp:291:106: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\nilus\Documents\Arduino\libraries\DS3231\DS3231.cpp:291:106: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\nilus\Documents\Arduino\libraries\DS3231\DS3231.cpp:291:106: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\nilus\Documents\Arduino\libraries\DS3231\DS3231.cpp:291:106: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\nilus\Documents\Arduino\libraries\DS3231\DS3231.cpp:291:106: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\nilus\Documents\Arduino\libraries\DS3231\DS3231.cpp:291:106: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

C:\Users\nilus\Documents\Arduino\libraries\DS3231\DS3231.cpp:291:106: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

Sketch uses 85206 bytes (33%) of program storage space. Maximum is 253952 bytes.
Global variables use 2917 bytes (35%) of dynamic memory, leaving 5275 bytes for local variables. Maximum is 8192 bytes.

closed 2021-02-13 by nilusha1993

bjrush, 2021-02-13:
I have been interested in building this for along time but have given up due to instructions on how to start. Bought tuns of supplies but the vidios were always going to be updated. I wanted to build a self sufficient hydroponics system for my 18x36 greenhouse. Have not worked on it in a year and a half. If you have info please i need help. Joe

Closed #129.

nilusha1993, 2021-02-14:
Hi joe, the system and files work, but its abit tricky to get it all placed in the correct places so the compiler doesnt error. anyways i got it all to compile using arduino 1.8.5 and working well now. email me on nilusha.iit@gmail.com for any problems you have ill help you out mate :)

bjrush, 2021-02-17:
Thanks for the info. But now that we have moved to the country I cant find any of my what I what I call plug and play schematics. I'll have to trank the websight I used so I can hook everything back up. Do you have any scamatics I can build from?

nilusha1993, 2021-02-18:
i followed drolsen schematics
https://raw.githubusercontent.com/drolsen/DRO-Matic/master/Docs/images/DRO-Matic_BreadboardSchematic.jpg

and watch the videos he explains how to wire.

bjrush, 2021-02-18:
Thanks I'll let you know how it goes. We are digging out of a bad ice and snow storm that hit 3 days ago here in Ardmore Oklakoma. Joe

https://raw.githubusercontent.com/drolsen/DRO-Matic/master/Docs/images/DRO-Matic_BreadboardSchematic.jpg